aster94/GoProControl #7: fixing last things

aster94, 2019-03-01:
@KonradIT 

let's discuss here the next steps
I just made two commit. In the first I moved all the defines in a new files, this way we have code and settings separated. I finally added support for esp32 but i also removed the support for not-esp boards. I decided to do so for many reason: first of all a esp board cost a few euros, there is no need to use AT commands and another MCU, second because the library for the AT commands doesn't support http request so with #5 and to be more precise with this [changes](https://github.com/aster94/GoProControl/pull/5/files#diff-1f55c289b92d0d995c923c12588db184R131) compatibility has been lost. I may add it in future if i would ever need it

Then i made a commit for formatting, since we both use VSCode i think it would be a problem to mantain the repo with the same code style

To do list
- [ ] simplify the udp request
- [ ] periodic keep alive on background thread
- [ ] get status
- [ ] parse status

KonradIT, 2019-03-01:
It's your repo, so you make the rules :)

Good that finally the ESP32 is supported!

Fine by me to delete Constants.h, I thought this would be easier for debugging and pushing to GitHub.

`http.end()` is for ending the HTTP request.

aster94, 2019-03-01:
> I thought this would be easier for debugging and pushing to GitHub

I completely agree, this way we don't have to overwrite each own SSID and password, we could keep it until we are heavily working on the lib, but I am pretty sure that some newby would find it not so easy 😅

I completed the first part of the rework:
- added the `_` before the private variables (it is a kind of c/cpp tradition)
- renamed some variables for readability
- added ability to change debug `Serial`
- updated gitignore (I may have done some error so check if it is working properly before pushing!)
- removed almost all the part of code about HERO and HERO2, i guess that almost no one have these camere nowadays

> http.end() is for ending the HTTP request.

yep the problem is that [that line](https://github.com/aster94/GoProControl/blob/master/src/GoProControl.cpp#L645) won't ever been read because there is a `return` before it, i think it should go after `_debug_port->println(payload);` tell me what do you think about
also really the variables `httpCode` and `payload` doesn't appear to be very useful because we don't use them so we should be able to delete them

if you was thinking to buy a non-esp board i could add again the code, tell me 😃

aster94, 2019-03-01:
okkkkk the settings are now reworked for a safer use! before it was possible to do `setOrientation(VR_4K);` now it would raise an error, I achieved it with a quite ugly part of code using enum but at least it works 😅

I added some nice looking tables in the readme!

I am having some doubts about the link between video resolution and frame rate: are them linked or not? from your repo i see that when i set a [video resolution](https://github.com/KonradIT/goprowifihack/blob/master/HERO3/WifiCommands.md#video-resolutions-black-edition) i set also a frame rate. But so what do i cange when i change directly the [frame rate](https://github.com/KonradIT/goprowifihack/blob/master/HERO3/WifiCommands.md#frame-rate)?

KonradIT, 2019-03-02:
Regarding video resolution and framerate;

- HERO3, 3+ cameras: Video Resolution will change the frame rate as well as per [the API](https://github.com/KonradIT/goprowifihack/blob/master/HERO3/WifiCommands.md#video-resolutions-black-edition) but you can also set a different Frame rate afterwards.
- HERO4 and newer: set video resolution and framerate separately.

aster94, 2019-03-02:
okkkk interesting! so i guess that in a HERO4 or later if i try to set the video resolution to 4k and the frame rate to 240 i would get a bad request or similar, correct?

KonradIT, 2019-03-03:
Perfect! 

>so i guess that in a HERO4 or later if i try to set the video resolution to 4k and the frame rate to 240 i would get a bad request or similar, correct?

Exactly

Regarding the MKR boards, [seems like `WiFiClient` is used](https://www.arduino.cc/en/Tutorial/Wifi101WiFiWebClient)

Regarding sending the keep alive command, seems it's not possible due to the board being single core, but we might be able to use the `loop()` function.

aster94, 2019-03-03:
> Regarding sending the keep alive command, seems it's not possible due to the board being single core, but we might be able to use the `loop()` function.

Exactly, it would be the easiest approach, really the esp boards could be used with freeRTOS but I don't know how to set it up and i suspect it wouldn't be so easy so I would definitely go for a simple `keepAlive()` function at the end of the `loop()`. What do you suggest to send to the gopro as command for this function?

> Regarding the MKR boards, seems like WiFiClient is used

Yesterday i tired to support these other boards but i faced few problems since the wifi API is standard but the http client is not (the HTTP functions unfortunately are different from esp* and other boards). Instead of using the `HTTPClass` of esp32 and esp8266 it would be better to use a different library that could work for all the client instances, like this one: https://github.com/arduino-libraries/ArduinoHttpClient 

Also I think that for best maintainability we should leave the user choose which wifi library should be used, for example (pseudocode):
```
#include <GoProControl.h>
#include <YourWiFiLibrary.h>
YourWiFiClass client
GoProControl(client, ssid, pwd, camera);
``` 
This way the user, depending on his hardware, could include the correct library and we wouldn't have to fight with #ifdef and precompiler flags to detect the correct board. Also, meanwhile new boards and wifi library are pubblished we won't have to change a single line of code and we could focus only on the new HERO commands

aster94, 2019-03-06:
@KonradIT now the http requests are based only on the lib `ArduinoHttpClient` soon the lib would be ready to run on all the boards.

do you know if there is a way to ask to the gopro if it is turned on?

how should be used the `comfirmPairing()` is it needed? with my camera i guess not

a few question about some features which i guess are specific about the hero4 or later: what is the sendWoL and the UDP? and how it is supposed to work? is it a wake on lane i guess but in the hero3 i could just send a command to do it 😅

also how should the bluetooth work with these camera? i could start writing some pseudocode to make it easier in the future to add it

KonradIT, 2019-03-07:
Hi, I have an ESP32 board and it's indeed faster.

- `confirmPairing()` is used to confirm a pairing for the first time on HERO5 and newer
- The mac address is needed to send the Wake on LAN command, this is to turn the camera on
- For newer camera: check if `http://10.5.5.9/gp/gpControl/status` can connect (HTTP 200), if not then  Wake on LAN -> then check again after a few seconds. if yes http code is 200 you can issue commands normally.
- I've been exploring bluetooth and wrote a [blog post](https://medium.com/@konrad_it/how-to-avoid-losing-your-gopro-with-around-15-7d53ac9bb609) about new BT capabilities in GoPro cameras. Basically there are two modes:
    - The Pairing mode: Since HERO5 cameras, when you want to connect a phone to it for the first time you need to put the camera in pairing mode, this starts a BT server with a few characteristics such as Camera Name, camera Password and a WRITE characteristic to send commands to. This BT server is only activated when the camera is in pairing mode, so it's a pain to use it (because you have to swipe, go to connection > new connection > GoPro App and then wait). But it's proven that it works well, you can [send GATT commands using `gattool`](https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/Platforms/RaspberryPi.md) ([list of commands here](https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/Platforms/ArchLinux.md)) and [`gatt-python` works relatively well with the camera](https://github.com/KonradIT/gopro-ble-py). You can try these codes with a HERO5 or newer camera
    - Undocumented BLE client mode: When GoPro had the developer program active they made a new mode for cameras which is Bluetooth mode, it's not really mentioned anywhere but it allows the camera to connect to an arbitrary server as soon as it turns on, and it will always try to connect when the Wireless mode (Wireless mode !== WiFi) is on. More about this mode on my blog post.

I've been trying to send GATT commands to the camera when it's on BLE client mode but cannot do it on the ESP32. Would be nice. As mentioned in the goprowifihack docs, we can turn off WiFi via bluetooth so the camera does not get warm and most importantly the battery lasts much longer.
- On WiFi: The camera needs a Keep Alive command to be sent every 2 seconds in order to stay on. Otherwise after a couple of minutes it turns off. Keep Alive command is shown here: https://github.com/KonradIT/gopro-py-api/blob/master/goprocam/GoProCamera.py#L70

I'll try out the ESP32 code. The ESP8266 does not support two threads due to it being single core right?

telemething, 2019-03-07:
Esp8266 runs RTOS and supports threads
https://github.com/espressif/ESP8266_RTOS_SDK

- Mark West

aster94, 2019-03-08:
> * `confirmPairing()` is used to confirm a pairing for the first time on HERO5 and newer

Great, that is why it wasn't working for my camera 😆

> * For newer camera: check if `http://10.5.5.9/gp/gpControl/status` can connect (HTTP 200)

Perfect, I will use it! That is a pity that I can't know if the hero3 is turned on or off 😫

> * I've been exploring bluetooth and wrote a [blog post](https://medium.com/@konrad_it/how-to-avoid-losing-your-gopro-with-around-15-7d53ac9bb609) about new BT capabilities in GoPro cameras. Basically there are two modes:

ok I would really love to do it even if doesn't seem to be so easy but I would like to suggest to leave it outside for the moment and focus on the basic workings, meanwhile i will buy a new camera 😆

> * On WiFi: The camera needs a Keep Alive command to be sent every 2 seconds in order to stay on. Otherwise after a couple of minutes it turns off. Keep Alive command is shown here: https://github.com/KonradIT/gopro-py-api/blob/master/goprocam/GoProCamera.py#L70

I will add it in the next commit, anyway is it possible that this is only on newer cameras? I am pretty sure that mine stays on

> I'll try out the ESP32 code. The ESP8266 does not support two threads due to it being single core right?

as @telemething said it supports it but (from our point of view) the technical answer is no, because the esp8266 arduino framework isn't based on freeRTOS, see [here](https://github.com/esp8266/Arduino/issues/2118)

aster94, 2019-03-08:
ok with this last push i almost finished with the structure of the library. I wrote some function for the BLE and adding support for it shouldn't be too hard but since i don't have any new gopro i can't do it

the library should work for all the board except for the boards which used the ESP01 with AT commands, I will check if i am able to do something for them

I didn't worked on the `confirmPairing()` for the HERO4, but now the HERO5 should work

most notably the new constructor: now you could pass to it the mac address and other parameters for the WoL

now the `isOn()` function should work with HERO4 and newer

same for the `keepAlive()` it should work, i didn't do nothing for the HERO3 because the connection stays alive for more then 6 minutes (i didn't test the maximum), it is enough in my opinion
```
15:39:06.600 -> Camera connected
15:39:06.600 -> Client connected
15:39:07.604 -> HTTP request: http://10.5.5.9/camera/SH?t=xxxxx&p=%00
15:39:07.639 -> Response: 200
15:39:07.639 -> Command: Accepted

15:46:51.718 -> Camera connected
15:46:51.718 -> Client connected
15:46:52.715 -> HTTP request: http://10.5.5.9/camera/SH?t=xxxxx&p=%01
15:46:52.715 -> Response: 200
15:46:52.749 -> Command: Accepted
```

as always i tested only on HERO3 and esp32 if you test them in your cameras please let me know, I am curios to know if the new stuff will work!

I think that i am not going to make any big changes soon so if you want to add more stuff feel free!

KonradIT, 2019-03-08:
Parse status means get the `http://10.5.5.9/gp/gpControl/status` JSON content and determine if the camera is recording, current mode, battery, how many minutes left of video/how many pictures left, resolution, fps...

aster94, 2019-03-08:
Okkk i thought that this was "get status"! Tell me when you have time to test the changes 😁

KonradIT, 2019-03-09:
Regarding the new changes... For WiFi, I think we should make it as easy as possible to deploy on ESP8266 and ESP32, so the GoProControl.h should detect which board is being sent to and include the appropiate libraries. Been trying to run the new changes to no avail.

KonradIT, 2019-03-09:
Because I cannot run the new changes on my ESP32 or ESP8266.

We should probably leave the  `WiFiClient gopro_client;
WiFiUDP udp_client;` to the GoProControl.h and set that up depending on the board. Constructor should look like this: 

`GoProControl gp( GOPRO_SSID, GOPRO_PASS, CAMERA, gopro_mac_address)`

KonradIT, 2019-03-09:
ArduinoHttpClient refuses to work: https://hastebin.com/ifimukopoy.sql

I'd say we go back to `HTTPClient` since it works with ESP32 and ESP8266.

aster94, 2019-03-09:
I have to admit that i also would prefer to have the library as much as possible plug&play, the idea to leave the user choose its library is nice but not everyone knows which library should they get

Anyway there is something strange, all the examples compile and run on my esp32 why is your not working? Which error do you get?

KonradIT, 2019-03-09:
Exactly, even I could not figure out exactly what library I needed at the first try, and even with the correct library installed (ArduinoHttpClient) it still did not work.

aster94, 2019-03-09:
I modified the constructor to the what you proposed + the name of the board (it is needed for the pairing https://github.com/aster94/GoProControl/blob/81cd28405f31ef4bf605e1b719c1611e5f93572d/src/GoProControl.cpp#L1557)

and tested on:
- ESP32
- MKR1000
- MKR WiFi 1010
- MKR VIDOR 4000
- UNO WiFi Rev.2
- Arduino UNO

the only board where it does not compile is the esp8266 due for a strange problem with their core, i am opening an issue right now to ask for suggestion

aster94, 2019-03-09:
I just made a test to check the wifi speed of my hero3+esp32:
time to run 10x shoot: 19:01:16.957 - 19:01:05.840 = ~11 seconds
how does you camera perform?
I made it with the arduino ide enabling the timestamp in the serial monitor and sending `AAAAAAAAAA` but you can do it with other terminal tool (not sure if it is possible in platformio)
 
```
19:01:05.840 -> Camera connected
19:01:05.840 -> Client connected
19:01:06.870 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:06.870 -> Response: 200
19:01:06.870 -> Command: Accepted
19:01:06.973 -> 
19:01:06.973 -> Camera connected
19:01:06.973 -> Client connected
19:01:07.962 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:07.996 -> Response: 200
19:01:07.996 -> Command: Accepted
19:01:08.064 -> 
19:01:08.064 -> Camera connected
19:01:08.064 -> Client connected
19:01:09.097 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:09.097 -> Response: 200
19:01:09.097 -> Command: Accepted
19:01:09.198 -> 
19:01:09.198 -> Camera connected
19:01:09.198 -> Client connected
19:01:10.225 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:10.225 -> Response: 200
19:01:10.225 -> Command: Accepted
19:01:10.326 -> 
19:01:10.326 -> Camera connected
19:01:10.326 -> Client connected
19:01:11.357 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:11.357 -> Response: 200
19:01:11.357 -> Command: Accepted
19:01:11.427 -> 
19:01:11.427 -> Camera connected
19:01:11.427 -> Client connected
19:01:12.454 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:12.454 -> Response: 200
19:01:12.454 -> Command: Accepted
19:01:12.558 -> 
19:01:12.558 -> Camera connected
19:01:12.558 -> Client connected
19:01:13.588 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:13.588 -> Response: 200
19:01:13.588 -> Command: Accepted
19:01:13.690 -> 
19:01:13.690 -> Camera connected
19:01:13.690 -> Client connected
19:01:14.689 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:14.724 -> Response: 200
19:01:14.724 -> Command: Accepted
19:01:14.792 -> 
19:01:14.792 -> Camera connected
19:01:14.792 -> Client connected
19:01:15.820 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:15.820 -> Response: 200
19:01:15.820 -> Command: Accepted
19:01:15.924 -> 
19:01:15.924 -> Camera connected
19:01:15.924 -> Client connected
19:01:16.957 -> HTTP request: http://10.5.5.9/camera/SH?t=catania94&p=%01
19:01:16.957 -> Response: 200
19:01:16.957 -> Command: Accepted
```

KonradIT, 2019-03-09:
Compiles on ESP32 finally! 

I get this now:

`19:45:26.829 -> 
19:45:26.829 -> Not connected
19:45:26.829 -> Camera not supported
`

I think the enums are not setting correctly, because `_camera` is 0.

aster94, 2019-03-09:
I am almost sure that this line is not working https://github.com/aster94/GoProControl/blob/d74c1badf05eba2b2d6015bc3ff5984c14558f2e/src/GoProControl.cpp#L53

copy this and replace that line:
```
_ssid = ssid;
	_pwd = pwd;
	_camera = camera;

	if (_camera == HERO3) // HERO3, HERO3+, HERO3BLACK, HERO3BLACK+
	{
		// URL scheme: http://HOST/param1/PARAM2?t=PASSWORD&p=%OPTION
		// example:	  http://10.5.5.9/camera/SH?t=password&p=%01

		_url = "http://" + _host + "/camera/";
	}
	else if (_camera >= HERO4) // HERO4, 5, 6, 7:
	{
		// URL scheme: http://HOST/gp/gpControl/....
		// Basic functions (record, mode, tag, poweroff): http://HOST/gp/gpControl/command/PARAM?p=OPTION
		// example change mode to video: http://10.5.5.9/gp/gpControl/command/mode?p=0
		// Settings: http://HOST/gp/gpControl/setting/option
		// example change video resolution to 1080p: http://10.5.5.9/gp/gpControl/setting/2/9)

		_url = "http://" + _host + "/gp/gpControl/";
	}
```

aster94, 2019-03-09:
goooooood news! the esp8266 was already working on the issue we found and it will be fixed soon https://github.com/esp8266/Arduino/issues/5749#issuecomment-465678078 thanks @jandrassy

jandrassy, 2019-03-09:
> goooooood news! the esp8266 was already working on the issue we found and it will be fixed soon [esp8266/Arduino#5749 (comment)](https://github.com/esp8266/Arduino/issues/5749#issuecomment-465678078) thanks @jandrassy

until then use esp8266 package 2.4.2

KonradIT, 2019-03-09:
Good! For some reason I get a 400 bad request 

`20:05:15.184 -> Not connected
20:05:15.184 -> Attempting to connect to SSID: KonradHERO7Black
20:05:15.184 -> using password: dude2903
20:05:15.184 -> Attempt: 1
20:05:15.316 -> WiFi.status(): 6
20:05:15.515 -> Attempt: 2
20:05:16.179 -> WiFi.status(): 6
20:05:16.378 -> Attempt: 3
20:05:17.041 -> Connection failed
20:05:19.695 -> 
20:05:19.695 -> Not connected
20:05:19.695 -> Attempting to connect to SSID: KonradHERO7Black
20:05:19.695 -> using password: dude2903
20:05:19.695 -> Attempt: 1
20:05:19.695 -> Connected to GoPro
20:05:24.107 -> 
20:05:24.107 -> Camera connected
20:05:24.107 -> Client connected
20:05:25.136 -> HTTP request: http://10.5.5.9/gp/gpControl/status
20:05:25.136 -> Response: 400
20:05:25.136 -> Command: Unknown error
20:05:25.169 -> Client connected
20:05:26.164 -> HTTP request: http://10.5.5.9/gp/gpControl/command/shutter?p=1
20:05:26.164 -> Response: 400
20:05:26.164 -> Command: Unknown error
`

By the way the keep alive should be every 2 seconds to avoid spamming the camera too much.

aster94, 2019-03-09:
mmmm I am almost sure that the problem is is keepAlive try to decomment this line:
https://github.com/aster94/GoProControl/blob/d74c1badf05eba2b2d6015bc3ff5984c14558f2e/src/GoProControl.cpp#L173
as i said you i don't know much about client request 😅

 
> until then use esp8266 package 2.4.2

just tested, works perfectly!

KonradIT, 2019-03-09:
comment or decomment? I've tried both.

aster94, 2019-03-09:
sorry i meant comment, if the problem isn't there i am not sure where it could be, the request string seems good, but the gopro says it isn't 🤔

KonradIT, 2019-03-09:
Yeah it's weird, no problem with HTTPClient.

KonradIT, 2019-03-15:
What are the advantage of using ArduinoHttpClient as opposed to HTTPClient? Since ArduinoHttpClient breaks connections with HERO5, 7 and possibly any HERO4 and newer camera.

aster94, 2019-03-16:
ArduinoHttpClient will work with any wifi library, so the pros are mainly:
1. maintainability
2. no repetition of the code to support different HTTP libraries
3. it is going to become part of the official arduino API (the same as the WiFi class) so adopting it would be better for the future

The best things would be to file an issue on ArduinoHttpClient but I don't have any idea how to help the arduino's developers to find and fix this bug 😕 if you have any idea on how to do this it would be perfect

If not I suggest, for now, to use the built-in HTTPClient in the ESP* boards and in the others the ArduinoHttpClient, I am quite sure i can do some #define preprocessor to use the two libraries at the same time without any repetition of the code

example:
```cpp
#if defined esp32 or esp8266
#define get GET
#endif
_http.get(request);
```
so one day (when and if the ArduinoHttpClient will fix the problem) we would just delete the define part and we would have all ready

aster94, 2019-03-17:
@KonradIT just for my own curiosity do you have any way to check what actually send the _http.get(request) using the two libraries?

anyway i was thinking that the simplest way to to do the http request would be to format them ourselves, no need to have any additional library, this should be enough
```
client.print("GET " + request + " HTTP/1.1\r\n");
client.print("Host: " + _host + "\r\n");
client.print("Connection: close\r\n\r\n");
```
Maybe it is even too much, since my camera worked without sending the host and *probably* the `Connection: close` is done by `client.stop();`

KonradIT, 2019-03-17:
Good news!

This works on my HERO7 Black and Fusion:

````

......

IPAddress server(10,5,5,9);
void setup() {
....
}
void loop() {

  if ((WiFi.status() == WL_CONNECTED)) { 
    Serial.println("sending");

    if (client.connect(server, 80)) {
      Serial.println("connected");
    String request = "/gp/gpControl/command/shutter?p=1";
    client.println("GET " + request + " HTTP/1.1");
    client.println("Host: 10.5.5.9");
    client.println("Connection: Keep-Alive");
    client.println();
    }
    

  }

  delay(1000);

}

````

KonradIT, 2019-03-17:
We can also read the response:

23:32:41.136 -> HTTP/1.1 200 OK
23:32:41.136 -> Content-Type: application/json
23:32:41.136 -> 
23:32:41.136 -> {}
23:32:41.136 ->

aster94, 2019-03-17:
Great 😃 so maybe also newer camera can use the `Connection: Keep-Alive`!

Just to be sure if you could test the comunication between your phone and the camera it would be perfect, tomorrow i would do all the changes to make the request from the library exactly the same to the one from the phone

KonradIT, 2019-03-17:
Oh and my camera definitively needs the constant thread for Keep Alive.

aster94, 2019-03-17:
I talked too early ahahah 😂

KonradIT, 2019-03-17:
Sure, using packet capture:

````
GET /gp/gpControl/command/shutter?p=1 HTTP/1.1
Host: 10.5.5.9
Connection: Keep-Alive
````

Response:

````
HTTP/1.1 200 OK
Content-Type: application/json

{}

````

aster94, 2019-03-17:
Okkk it is almost the same but the host is without the port (my app add ":80")

Are you able to see when the app sends the "_GPHD_:0:0:2:0.000000"? It could be interesting to check the formattatin of this message

KonradIT, 2019-03-17:
Yes, It's the same as we have it right now.

aster94, 2019-03-23:
done! All is working good in my camera, if in yours everything is ok I think we are ready to close the issue!

KonradIT, 2019-03-23:
Nice I'll give it an overall test tomorrow.

KonradIT, 2019-03-24:
2 problems still needing to be solved:

1) `13:52:58.821 -> E (99) FreeRTOS: FreeRTOS Task "keep_alive" should not return, Aborting now!` in ESP32 RTOS .ino

2) `gp.keepAlive()` stalls server. I think if we ping `/gp/gpControl/status` every 5 seconds it should dothe job.

aster94, 2019-03-24:
#9 merged!

I am not really sure aboutn the error in freertos since i never used it, i wrote that sketch using this tutorial: https://techtutorialsx.com/2017/05/06/esp32-arduino-creating-a-task/

and googling the error i found https://www.freertos.org/implementing-a-FreeRTOS-task.html

so maybe adding `vTaskDelete( NULL );` at the end of the task should work, maybe!

aster94, 2019-03-25:
#10 merged! I will make a small review of the code and clean a few function, especially i will rework the listen function and remove the String() class since it can bring to memory corruption, i added what you proposed to the todo list in the first post but, honestly, not having a camera it wouldn't be so easy for me 😅

aster94, 2019-03-26:
finishhhhhhh

so I made minimal changes to the code, mainly i edited the  `listenResponse()` and small fixes around

I had to comment out the code for the BSSID because there was an issue of compatibility with the arduino API https://github.com/espressif/arduino-esp32/issues/2613

I added a section in the readME for the next steps, if you want to make any of them https://github.com/aster94/GoProControl#to-do-list-and-known-issues

I wanted to ask you if [this](https://github.com/KonradIT/goprowifihack/blob/master/HERO4/Framerates-Resolutions.md) applies for all HERO4,5,6,7 and fusion cameras or is only for hero4? because in your repo the hero5 and newer point here

KonradIT, 2019-03-29:
>I wanted to ask you if this applies for all HERO4,5,6,7 and fusion cameras or is only for hero4? because in your repo the hero5 and newer point here

Technically yes, HERO6, 7 have double the framerates on each resolution except 720p, 960p and WXGVA.

Regarding pairing, it should be done when the camera is bought and turned on for the first time, this is to kick it into AP mode. This can also be done via the smartphone App.

aster94, 2019-03-30:
> Regarding pairing, it should be done when the camera is bought and turned on for the first time, this is to kick it into AP mode. This can also be done via the smartphone App.

Great since the pairing process could be done from the smartphone app i would avoid passing the `board_name` field all the times. At the end the constructor could be without the `mac_address` and `board_name` fields, very neat!

KonradIT, 2019-03-30:
Yes even better.

aster94, 2019-04-14:
done 😃 now the mac_address is not needed anymore! except for the esp8266, because they have a limitation with the SDK (https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#bssid)

if you have the time to test the library tell me if it works! I didn't updated the version so you would need to manually download it, when you will confirm that all works i will update it

KonradIT, 2019-04-15:
Great! I'll test this afternoon

KonradIT, 2019-04-16:
Commit coming now that I tested everything on ESP32. Turing on does not seem to work yet when getting MAC address from camera via the board.

KonradIT, 2019-04-22:
Re: isOn(), I don't have a HERO3 anymore but I remember /bacpac/sd or /camera/sx or /camera/se not responding when the camera was off.

aster94, 2019-04-22:
good trancks! I will test this tomorrow!

aster94, 2019-04-25:
I run a few test:
all these are with camera connected but off:

- take a picture
```
HTTP/1.1 410 Gone
Allow: GET 
Accept-Ranges: bytes
Cache-Control: no-cache
Cache-Control: no-store
Connection: Keep-Alive
Server: GoPro Web Server v1.0
Content-Type: application/octet-stream
Content-Length: 1


```

- /bacpac/sd
always respond with 200

- camera/sx
```
HTTP/1.1 403 Forbidden
Allow: GET 
Accept-Ranges: bytes
Cache-Control: no-cache
Cache-Control: no-store
Connection: Keep-Alive
Server: GoPro Web Server v1.0
Content-Type: text/plain
Content-Length: 15

403 Forbidden
```

- /camera/se
```
HTTP/1.1 403 Forbidden
Allow: GET 
Accept-Ranges: bytes
Cache-Control: no-cache
Cache-Control: no-store
Connection: Keep-Alive
Server: GoPro Web Server v1.0
Content-Type: text/plain
Content-Length: 15

403 Forbidden
```

The weird part was then from the last two request I always get a reboot of the chip and i really can't understand from where they came... I lost the half hours trying to figute them out with no luck, this is the output i got
```
Response code: 403

Stack smashing protect failure!

abort() was called at PC 0x400dbfc4 on core 1

Backtrace: 0x4008f6f4:0x3ffb1e80 0x4008f8f7:0x3ffb1ea0 0x400dbfc4:0x3ffb1ec0 0x400d235e:0x3ffb1ee0 0x400d2456:0x3ffb1f20 0x400d2707:0x3ffb1f50 0x400d19fb:0x3ffb1f80 0x4011f802:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
```

closed 2020-03-24 by aster94