aster94/GoProControl #14: Cant conect my camera

kennyg100, 2019-06-13:
Cant connect to my camera - always the same error 
After C command -
Attempting to connect to SSID: ALLGP001
using password: bike7632
....................
Connection failed with status: 6

My Constants.h - 

#ifndef CONSTANTS_H
#define CONSTANTS_H

// replace the following:
#define GOPRO_SSID "ALLGP001"
#define GOPRO_PASS "bike7632"
#define CAMERA HERO5 // your HERO model
uint8_t gopro_mac_address[6]= {246,221,158,229,124,140}; // the MAC address of your GoPro

#endif

Board 
Node MCU v3

What can I do ?
Regards - Robert

aster94, 2019-06-13:
Hi Robert, usually i get the error 6 the first time i press 'C' but if i try again it works, have you tried to press it a few times and see what changes?
Also have you tried to use that user and password with your phone and see if that works?

kennyg100, 2019-06-14:
Thank you very much for fast answer , but repeat C command change nothing. maybe my Node MCU board is broken , but i check that hardware as demo server and everything working good. Thank You again.
Best regards
Robert

aster94, 2019-06-14:
mmm honestly i don't know, you could try to reset the firmware of your esp8266, try a different board, or even better an esp32

Strat00s, 2019-08-23:
Hey. Just found out about this library, Just tried it and I have the exact same problem as you. It will always (no matter how many times I try to connect) fail with "Connection failed with status: 6"
Is there any conclusion/solution to this behavior?
Here is my setup:
```
#define GOPRO_SSID "GP26220945"
#define GOPRO_PASS "wave6273"
#define CAMERA HERO7 // your HERO model
uint8_t gopro_mac_address[6]= {0x06, 0x41, 0x69, 0xd8, 0x49, 0x63};
```
I also tried it DEC address, but that did not change anything. Failes every tme.
`uint8_t gopro_mac_address[6]= {6, 65, 105, 216, 73, 99};`

aster94, 2019-08-25:
Using same ssid and pass with a mobile phone and gopro app works?

Strat00s, 2019-09-04:
> Using same ssid and pass with a mobile phone and gopro app works?

Yup. But in the end,  I achieved what I needed with a simple HTTP request.

closed 2020-03-24 by aster94