aster94/GoProControl #15: No BSSID, unable to turn on the camera

hendrikjvr, 2019-07-18:
Hi,

Trying to run your GoProControl example on an ESP32 with a GoPro Hero5 Session. It seems to work, except for being able to turn on the camera by sending a 'T' over the serial monitor. I get this response back:
--------------------------------------------------
Not connected
Attempting to connect to SSID: GP56217423
using password: action2315
..................
Connected to GoPro
Keeping connection alive
Client connected
Request: _GPHD_:0:0:2:0.000000

Keeping connection alive

Camera connected
No BSSID, unable to turn on the camera
--------------------------------------------------
To me it seems like the MAC address is not read? I can see that it is set to NULL in the constructor, but where is it changed or read from the camera?

aster94, 2019-07-18:
The Mac is obtained calling `getBSSID() `, which is called while you turn off your camera

hendrikjvr, 2019-07-18:
So you have to manually power on the gopro the first time?

aster94, 2019-07-18:
Yes, or you could pass it to the constructor. If you want to make a upgrada to the code to solve this feel free to make a pull request

aster94, 2020-03-27:
now the BSSID can be obtained during the connection

closed 2020-03-27 by aster94