aster94/GoProControl #17: support Hero 8

radimsejk, 2019-11-22:
Hello, is it planned to add support for Hero 8 camera and when if so? Thanks.

KonradIT, 2019-11-22:
Ah, I have the MAX, and working to get BT specs. Once the Hero 8 Black I will also add support.

aster94, 2019-11-23:
> Hello, is it planned to add support for Hero 8 camera and when if so? Thanks.

obviously yes but since i don't have any of them i won't be able to test. if you have one of them you can try to make the changes and make a pull request

radimsejk, 2019-11-25:
I have Hero 8 and I am willing to help. But I do not know how. Can you quide me what should I do?

aster94, 2019-11-25:
just use the normal example assuming your gopro is a hero7, let's see how it works, I didn't had a look to the new gopro and i don't know if the pairing sequence changed

radimsejk, 2019-11-26:
Well that is the problem. I have tried Hero 7 and it does not work. When I try to connect it always ends with "Connection failed with status: 6" message.

KonradIT, 2019-11-26:
What board are you using?

radimsejk, 2019-11-26:
ESP8266 (Wemos D1 mini Pro)

KonradIT, 2019-11-26:
Weird, try setting the network to 2.4GHz.

radimsejk, 2019-11-26:
I have it set to 2.4 GHz (because I do not know why but it does not work with 5 GHz and my LG v40). I have also noticed that the SSID is not available until I try to connect with my phone app. It looks like it is in power saving until phone (I assume with bluetooth) will try to wake it.

KonradIT, 2019-11-26:
The HERO8? Yeah, you need to connect it to the app to turn on the WiFi AP.

radimsejk, 2019-11-26:
Yeah, but even when I am conencted to the app the Wemos says "Connection failed with status: 6"

radimsejk, 2019-11-26:
I have tried to connect multiple times after each other and after 12 tries it connected for a couple of seconds and than "Connection lost".

radimsejk, 2019-11-26:
Now it connects every time but with "Connection lost" afterwards. I can not get it do start video or take a picture in this interval (it is something around 2 seconds)

radimsejk, 2019-11-26:
When I try to make a web request from PC (connected to the GoPro Wifi and open`http://10.5.5.9/gp/gpControl/command/system/locate?p=1`) it starts beepeing. When I try to make webrequest (with [this example ](https://esp8266-shop.com/blog/how-to-http-get-and-post-requests-with-esp8266/) GET) it connects to WiFi, gets an IP address but the http request fails and no beeping occurs.

radimsejk, 2019-11-26:
Well, it looks like my Wemos board was broken. I have tried another one with exact same code and it works like a charm. Sorry for bothering you and thank you.

aster94, 2019-11-26:
Do not worry the wemos d1 is a shitty board, I try to don't use it

I am quite happy that the gopro 8 works without any changes 😁
Tonight I will make some minimum changes to the library, idk if the new camera have new functionality compared to hero7, @KonradIT do you know something?

KonradIT, 2019-11-26:
Yes there are some new stuff regarding modes, unlike HERO7. MAX documentation is up on goprowifihack and it shares a lot in common with HERO8 so we can use that as a starting point until I get the HERO8.

aster94, 2020-03-24:
So I am 100% sure that there are some issues with newer gopro, at least in the MAX, because they changed some configuration, for example the FOV ([before](https://github.com/KonradIT/goprowifihack/blob/master/HERO7/HERO7-Commands.md#fov), [after](https://github.com/KonradIT/goprowifihack/blob/master/MAX/MAX-Commands.md#lens-videotimewarp)). But they are very silly issues and very easy to solve so we cna say that we have support for HERO8 and MAX

closed 2020-03-24 by aster94