aster94/GoProControl #20: Trying to get the media list

roberto1963, 2019-12-20:
Hello, 
Congratulations for your works about gopro control, it works ok for me and I had a lot of fun with it !

But I would need a way to ensure that the pictures are really made when taking a photo. So I tried to modify the code to get the media list using this command : http://10.5.5.9:8080/gp/gpMediaList
I would parse the response to count the pictures.

Unfortunatly, my knowledge about arduino libraries is too weak to do the job , I didn't even succeed in compiling the modified library  ( "multiple definition" errors ) ...  8-(( 

In addition, I am not sure my idea would works :
_request = "http://10.5.5.9:8080/gp/gpMediaList" ; 
String Response=sendHTTPRequest(_request);
Would it be possible to do something like this ?

Many thanks for your help .
Roberto

aster94, 2019-12-21:
Ciao roberto,

what are you asking, i guess, is feasible. Can you post here the response you get before and after taking a picture from the `http://10.5.5.9:8080/gp/gpMediaList` command?

roberto1963, 2019-12-22:
I finally succeed to modify your library, and I have added this small code :
uint8_t GoProControl::mediaList() {   
 _request = "/gp/gpMediaList" ;     
  return sendHTTPRequest(_request);   }

When ESP32 call mediaList ,  it make EPS32 to reboot ...
Backtrace: 0x4008c7e0:0x3ffb1bf0 0x4008ca11:0x3ffb1c10 0x400dcd00:0x3ffb1c30 0x400d2efa:0x3ffb1c50 0x400d2ff7:0x3ffb1c90 0x400d32fd:0x3ffb1cc0 0x400d208e:0x3ffb1cf0 0x400d581f:0x3ffb1fb0 0x4008877d:0x3ffb1fd0
Rebooting...

when I try http://10.5.5.9:8080/gp/gpMediaList with my browser , answer is of gopro hero 3 is:
  404 Not Found
The requested URL /videos was not found on this server. 
Cherokee web server 1.2.101b121212_5bf2ec1 (UNIX), Port 80 

In browser , http://10.5.5.9:8080/ gives the Ambarella home page of gopro .

Trying shoot command gives : 403 Forbidden

aster94, 2019-12-22:
Ok, if you are able to make it working I will add to the library, unfortunately I don't have time to focus on this

KonradIT, 2019-12-22:
I'll give it a spin, ideally this would interface with whatever Arduino library can write to SD cards as well.

aster94, 2019-12-22:
That is external to the library, however, when it will work, an example could be written

roberto1963, 2019-12-24:
This request (http://10.5.5.9:8080/gp/gpMediaList) is it ok for gopro hero 3 ?
It doesn't seems to works for me when I test it with my brower ... why ?

KonradIT, 2019-12-24:
Weird, it worked on my HERO2 and HERO3 when I had them.

roberto1963, 2019-12-24:
So I made a error in my test ...
I will made other tries as soon as possible.
Using a browser is it the easiest solution to test this ?

aster94, 2020-03-24:
@roberto1963 done, now there is a funtion called `getMediaList()` check it from the example.
However I choosed to do not implement a json parser for the status and for the media list because the library is already enough memory hungry

closed 2020-03-24 by aster94

roberto1963, 2020-03-26:
Great !!!
Many thanks !
I will test it in the next days . 
I will try to write a small piece of code to get the number of pictures , so I would be able to check if a picture is done or not .

aster94, 2020-03-26:
great, i will be happy to integrate it in the examples, i would suggest you to have a look at this code https://arduinojson.org/v6/example/parser/

aster94, 2020-03-27:
@roberto1963 I made a lot of improvements, have a look at the arduinoJson example. It is working everything but to get the number of picture/video taken you will need to discover a little more how arduinojson manages the json files

This is an example from `getMediaList()`
```
{"id":"240009376192920323","media":[{"d":"100GOPRO","fs":[{"n":"GH010043.MP4","cre":"1585310042","mod":"1585310042","glrv":"1271358","ls":"-1","s":"14389838"},{"n":"GH010044.MP4","cre":"1585310152","mod":"1585310152","glrv":"2568816","ls":"-1","s":"27142182"},{"n":"GH010045.MP4","cre":"1585311234","mod":"1585311234","glrv":"3620178","ls":"-1","s":"40970466"}]}]}
```

every picture is one of these blocks: `{"n":"GH010043.MP4","cre":"1585310042","mod":"1585310042","glrv":"1271358","ls":"-1","s":"14389838"}` so in the above example there are 3 pictures, I didn't find a way to count them
you may wish to ask to someone who knows that library. I will wait to your response to update the example

roberto1963, 2020-04-05:
Hello Aster94,
I have made many tries since some days but I didn't succeed to get a medialist json message with my gopro hero 3 :
Here what I receive :

Start response body
HTTP/1.1 ⸮⸮Allow: GET 
Accept-Ranges: bytes
Cache-Control: no-cache
Cache-Control: no-store
Connection: Keep-Alive
Server: GoPro Web Server v1.0
Content-Type: text/plain
Content-Length: 2
End response body

The json message is it in _response_buffer array ?

I would like to search for .JPG occurences in json message to get the number of pictures.

Roberto

aster94, 2020-04-05:
roberto, the response you are receiving doesn't seem correct, have you tried to send a few times the request and did you get always this answer?

`HTTP/1.1 ⸮⸮Allow: GET` this is indeed quite strange

> The json message is it in _response_buffer array ?
yes, also, but it is returned by the getMediaList() which in your case won't return nothing because there is any media list