aster94/GoProControl #21: setTimeLapseInterval error ?

roberto1963, 2019-12-22:
Hello , 
Calling setTimeLapseInterval return a error when I call it (wrong parameter).
I think it is necessary to replace OR operations by AND operations in this line:

https://github.com/aster94/GoProControl/blob/2323f381f97ec14b305f264e04e9dae75d7ed8ad/src/GoProControl.cpp#L1095

When I change this line in :   
if (i_option != 0 && i_option != 1 && i_option != 5 && i_option != 10 && i_option != 30 && i_option != 60)
it works OK. 
Do you think it is ok ?
Roberto

aster94, 2019-12-22:
Can you post an extract of the code you are using?

roberto1963, 2019-12-24:
I only change all the || with && in the line 1095 and it works .

Here is the way I call timelapse mode :

  gp.begin ();
  gp.begin ();
  gp.turnOn();
  gp.setTimeLapseInterval(1);
  gp.setMode(TIMELAPSE_MODE);   
  gp.shoot();

aster94, 2020-03-24:
I checked this only now and you were absolutely right|

closed 2020-03-24 by aster94