aster94/GoProControl #22: Sleep Command

mattkerrison, 2020-04-21:
I've been trying out the GoProControl example, however I noticed that when you issue the turn off command the keep alive ping fails and can't connect again.

From reading Konrad's command list, it looks like it should just sleep:

_Power Off: http://10.5.5.9/gp/gpControl/command/system/sleep_

I would've thought it would continue to accept the keep alive pings, is this expected behaviour?

aster94, 2020-04-21:
Hi,

actually I am not sure, you can try to comment this line:
https://github.com/aster94/GoProControl/blob/4e63a60d4656224c5bb0b67eba2d7a94dc1ef8a6/src/GoProControl.cpp#L155
and check what happens, please let me know!

mattkerrison, 2020-04-21:
I thought it was because I was using an ESP8266:

> ESP8266 can't get the BSSID of the camera. So if you want to turn it on you need to manually pass it to the constructor, see the docs

However I just flashed an ESP32 and having the same issue, I'll try commenting out that line

aster94, 2020-04-21:
which camera are you using?

mattkerrison, 2020-04-21:
Hero 6 Black

aster94, 2020-04-22:
so the gopro is not answering to the keep alive command, have you tried to pass to the constructor the MAC address of your gopro? Here there is the first tutorial i ve gfound on how to get it (check if it is good your your camera): https://havecamerawilltravel.com/gopro/gopro-mac-address/

and here is where you should write it: 
https://github.com/aster94/GoProControl/blob/4e63a60d4656224c5bb0b67eba2d7a94dc1ef8a6/src/GoProControl.h#L76-L80

so for example:

```
uint8_t mac[] = {1,2,3,4,5,6}
GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA, );
```

closed 2021-09-04 by aster94