aster94/GoProControl #23: Hero7 Silver

dushyantahuja, 2020-04-28:
Just found this library, and loved the concept. However, installed the GoProControl.ino example and tried to make it work. I updated the Secrets.h file, but was not able to connect to the Hero7 Silver at all. KonradIT mentions that the calls work with the Hero7 Silver https://github.com/KonradIT/goprowifihack/tree/master/HERO7 -so not sure what I'm doing wrong. Any help would be appreciated.

dushyantahuja, 2020-04-28:
To try to debug this, I connected to the GoPro's wifi using my computer and opened the links in KonradITs documentation. They work, but the ESP8266 is simply not able to connect. I even changed the WiFi of the camera to 2.6 GHz - but still didn't work.

KonradIT, 2020-04-28:
We need log details from the ESP8266, to understand why the HTTP requests are not getting through.

dushyantahuja, 2020-04-28:
Closing this now - I was an idiot, and didn't realise that I had to click 'b' first to 'begin' the connection. It's working now - thanks a lot.

closed 2020-04-28 by dushyantahuja

aster94, 2020-04-28:
Set the wifi to 2.4GHz is mandatory since the esp8266 doesn't handle 5GHz.
as kondard said please paste a log of the serial with debug enabled

aster94, 2020-04-28:
never mind, have fun with the library, be aware that the ESP8266 have some limitations