aster94/GoProControl #24: Constants.h: No such file or directory

maxholgasson, 2020-06-03:
Hi,
I installed the lib and getting the error Constants.h: No such file or directory
Where do I get that file from?

closed 2020-06-03 by maxholgasson

aster94, 2020-06-03:
now it should be called `Secrets.h`

maxholgasson, 2020-06-03:
Thanks, yes I figured out that I had an older installation also on my computer.
However it's still not running
`Mehrere Bibliotheken wurden für "WiFi.h" gefunden
/Users/birnbaum/Documents/Arduino/libraries/GoProControl/src/GoProControl.cpp:23:23: fatal error: Utilities.h: No such file or directory
 Benutzt: /Users/birnbaum/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
compilation terminated.
 Nicht benutzt: /Applications/Arduino1812.app/Contents/Java/libraries/WiFi
exit status 1`

aster94, 2020-06-03:
you can download the utilities from the arduino/platformio lib manager, about the wifi.h conflicts that is a problem on some installation where you are using both arduino wifi boards and espressif

maxholgasson, 2020-06-03:
I'm using the classic arduino ide. is that platformio something different?

aster94, 2020-06-03:
which board are you using?

maxholgasson, 2020-06-03:
MH ET LIVE ESP32DevKIT
http://esp32.net/images/MH-ET-LIVE/ESP32-DevKit/MH-ET-LIVE_ESP32-DevKit_PhotoTopPerspective.jpg

aster94, 2020-06-03:
ok go to the Arduino lib manager and uninstall these libraries:
WiFi101, WiFiNINA, VidorPeripherals, WiFiEsp

this should solve the problem

maxholgasson, 2020-06-03:
I have non of these installed
Can I install the missing Utilities.h also for the classic arduino IDE?

aster94, 2020-06-03:
yes

maxholgasson, 2020-06-03:
Ok found it online as a zip and I could upload the sketch.
but can't connect to my Hero8
>Connection failed with status: 1

Is there a manual with error codes or how to pair it etc?

My camera ssid and pw are correct

aster94, 2020-06-03:
modern cameras need to be paired with BLE before wifi so at first you need to turn it on with the mobile phone and then you can connect it with your arduino
also the gopro must use the 2.4ghz band not 5ghz from the option of the camera

maxholgasson, 2020-06-03:
Ok. For my usage I need to cut the power connection, so I need to turn the camera on and start recording remotly without any smartphone..this I assume is not possible with this lib?

aster94, 2020-06-03:
you need to connect the camera with the phone, then you can use your arduino. the espressif BLE library is not enough stable to implement it yet. But you can always try and see if the esp32 crashes

maxholgasson, 2020-06-03:
Ok, but I have to use my phone everytime I cut off the power of the gopro, right?

aster94, 2020-06-03:
yes, unfortunately yes for now

maxholgasson, 2020-06-03:
Ok thanks, that's a pity. For my application I need to start recording without any person near the camera

aster94, 2020-06-03:
if you have some programming experience you can try to use the BLE, for example: https://github.com/KonradIT/gopro32/blob/master/examples/guardian.ino you should connect via BLE then send the command to turn on the WiFi `17, 01, 01`

The reason behind this is that wifi consumes a lot of power so gopro made this "complex" to turn the wifi off when possible. If the wifi was always on it would be far away easier

maxholgasson, 2020-06-03:
Thanks, I will  have a look at it. Ah ok yes I understand