aster94/GoProControl #25: I am trying, but I cant connect the camera

edsonchuck, 2020-06-13:
Hi Aster,

First I have to thank you for your amazing working on this project,

I am not a very experienced arduino programmer and I am trying  to connect a ESP8266 NodeMcu board to my Gopro Hero6 Black.

I followed all your instructions installing the library on my Arduino IDE, and I tried to use this example, but I can not make it works, and I have no idea why.

I just copy and paste the code, changed the Secrets.h with my camera SSID and Password, and changed the beginning of the code "CAMERA" for "HERO6". After that, I uploaded the code in the board, and until now, everything is ok. Then, I open the Serial Monitor, and try to use the commands in the code ("b", "c", etc), but nothing happen, and nothing appears in the Serial Monitor. In the camera not happens as well.

Any idea what I am doing wrong?

Thank you very much

aster94, 2020-06-13:
Hi, if nothing appear in the serial monitor and the debug is enabled something may be wrong in your setup

edsonchuck, 2020-06-14:
Thank you for your quick answer. 

I tried again, carefully copying your code, and nothing happens again.

So, I just uploaded a simple led blink code on the board, and it works fine.

Definitely I have no idea whats is going on. I'm gonna try a different code, with other wifi connections and see what happens.

The only thing, I don't know if I am doing right, is the Secrets.h file. I put there the camera SSID and password. But there is no place to put the camera model. So I put the camera model in the main code.

Like this: GoProControl gp(GOPRO_SSID, GOPRO_PASS, HERO6);

Is it ok doing like this?

Many thanks again

aster94, 2020-06-15:
If you want to can also get rid of `Secrets.h` and write your wifi name and password directly in the main sketch like this:
`GoProControl gp("my_ssid", "my_pass", HERO6);`

Still, I can't understand why you don't see any serial plotting

edsonchuck, 2020-06-26:
Hello, aster,

Finally, I solved the problem, and I am already trying to use the code in my own project.

After some tests, I am having a new problem. I can connect the camera, and everything works fine, but always after around 8 minutes, the board loses the connection with the camera. I am using the command "gp.keepAlive", but the connection closes anyway.

Do you have any idea about how the connection can stay always turned on?

I am trying to use the camera attached to a PIR movement sensor, so, the idea is the camera stay always turned on and start recording when any movement is detected.

Thank you very much again.

aster94, 2020-06-26:
maybe you are not calling the keep alive enough often. I have been using the lib for even 20minutes with very fast loops and i never had a problem, your code probably hangs somewhere and the function isn't called

closed 2021-09-04 by aster94