aster94/GoProControl #26: Unable to connect Hero 8

willemcvu, 2020-06-14:
My goal is to control a fleet of Hero 8 Black cameras with more reliability and control than the GoPro smart remote. Right now I'm just trying to control a single camera.

I've changed the GoPro's band to 2.4GHz, and entered the camera name and password in the `GOPRO_SSID `and `GOPRO_PASS `fields in secrets.h respectively.

With the `GoProControl.ino` example, when connecting via `b` command, the connection times out and I get a `Connection failed with status: 6` message.

I've tried on both an ESP8266 module as well as an ESP32 module.

Has this library been tested on a Hero 8, or just speculated working (based on the API being compatible with the Hero 8)?

Any help debugging this would be appreciated. Thanks!

willemcvu, 2020-06-14:
Quick update: I didn't realize that with the Hero 8, the phone app requests the camera to turn on WiFi via Bluetooth. This is now working great if I start the app, go to the "preview" window which forces the camera to enable WiFi. Then I can connect to the camera with the ESP32 or ESP8266 fine and fully control it as expected.

Perhaps with the ESP32, there would be value in reverse-engineering the bluetooth exchange by which the app enables WiFi, and then we can do the whole pairing process directly from the ESP32. There's a small discussion on BT here: https://github.com/KonradIT/gopro-py-api/issues/122

willemcvu, 2020-06-14:
Ah, I see you have this listed in the "known issues". Closing.

closed 2020-06-14 by willemcvu

aster94, 2020-06-15:
Hi Willem,

Yes unfortunately the wifi is turned on via BLE. The BT code is now maintained by the ESP team (https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE), something can be tried but I don't have a lot of experience with Bluetooth. This is the request to turn the Wifi on: 17, 01, 01

willemcvu, 2020-06-15:
Thanks for giving me the ble request to turn on the wifi, but I'm wondering where you found that out - I looked for ble documentation and most of what I could find was some logs with little analysis. Wondering if you found that out somewhere or did you reverse-engineer that yourself?

Thanks!

KonradIT, 2020-06-15:
http://github.com/konradit/goprowifihack

aster94, 2020-06-15:
Yes Willem, all commands came from Konrad's repo

maxholgasson, 2020-07-10:
Will there ever be a solution to wake up the camera and start recording via BLE? Or is this just not possible due to the hardware? I need to have the camera off, very low power for a long time and be able to wake it up when a trigger happens. At the moment I soldered a wire to the Hero8 trigger and using quikcapture. that works but it's a lot work to modify the cam and I don't get the status to verify if it's recording