aster94/GoProControl #27: Problem compiling GoProControl_Example

lacomj, 2020-06-25:
Hi,
First off, thanks for your support of the Arduino community!
I am having an unexpected problem simply compiling the example code (GoProControl).  I am getting an "invalid conversion" error at the call to the 'GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA);' line.  The error message(s) are shown below.

For context, this is using IDE version 1.8.13 on a Mac with an MKR 1010 WiFi.   The MKR board otherwise works fine (including WiFi features) with other code.  Some observations:

- I am using the current version of the GoProControl library (1.2.0), although I get similar errors with older versions of the library.

- I tried directly entering the SSID, and password into the main code (rather than using _Secrets.h_) but get errors for the same line (although the errors change)  I.e., `GoProControl gp("GP24907984", "cHS-Yy9-9Gv", HERO8);`

Any thoughts on what the problem (or solution) is for this conversion problem?
Thanks!
 
**Error Messages:**
> Arduino: 1.8.13 (Mac OS X), Board: "Arduino MKR WiFi 1010"
GoProControl_Example:16:47: error: invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]
 GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA);
                                               ^
In file included from /Users/lacomj/Documents/Arduino/GoProControl_Example/GoProControl_Example.ino:7:0:
/Users/lacomj/Documents/Arduino/libraries/GoProControl/src/GoProControl.h:76:3: note:   initializing argument 3 of 'GoProControl::GoProControl(const char*, const char*, uint8_t, const uint8_t*, const char*)'
   GoProControl(const char *ssid,
   ^~~~~~~~~~~~
exit status 1
invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

aster94, 2020-06-26:
This is indeed strange, can you try to do what is suggested here:

```
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

the problem seems to be about the `camera`: `initializing argument 3 `

lacomj, 2020-06-26:
Sure, here is the more verbose (details) output.  This is pretty long.  The "good stuff" (which was shown in the less verbose version) is at the bottom.

> Arduino: 1.8.13 (Mac OS X), Board: "Arduino MKR WiFi 1010"
/Applications/Arduino v1.8.13.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino v1.8.13.app/Contents/Java/hardware -hardware /Users/lacomj/Library/Arduino15/packages -tools /Applications/Arduino v1.8.13.app/Contents/Java/tools-builder -tools /Applications/Arduino v1.8.13.app/Contents/Java/hardware/tools/avr -tools /Users/lacomj/Library/Arduino15/packages -built-in-libraries /Applications/Arduino v1.8.13.app/Contents/Java/libraries -libraries /Users/lacomj/Documents/Arduino/libraries -fqbn=arduino:samd:mkrwifi1010 -vid-pid=2341_8054 -ide-version=10813 -build-path /var/folders/yp/z8lxwl6n16vgzh5yqzx7ztnw0000gn/T/arduino_build_475826 -warnings=none -build-cache /var/folders/yp/z8lxwl6n16vgzh5yqzx7ztnw0000gn/T/arduino_cache_727838 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.CMSIS.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arduinoOTA.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.openocd.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -verbose /Users/lacomj/Documents/Arduino/GoProControl_Example/GoProControl_Example.ino
/Applications/Arduino v1.8.13.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino v1.8.13.app/Contents/Java/hardware -hardware /Users/lacomj/Library/Arduino15/packages -tools /Applications/Arduino v1.8.13.app/Contents/Java/tools-builder -tools /Applications/Arduino v1.8.13.app/Contents/Java/hardware/tools/avr -tools /Users/lacomj/Library/Arduino15/packages -built-in-libraries /Applications/Arduino v1.8.13.app/Contents/Java/libraries -libraries /Users/lacomj/Documents/Arduino/libraries -fqbn=arduino:samd:mkrwifi1010 -vid-pid=2341_8054 -ide-version=10813 -build-path /var/folders/yp/z8lxwl6n16vgzh5yqzx7ztnw0000gn/T/arduino_build_475826 -warnings=none -build-cache /var/folders/yp/z8lxwl6n16vgzh5yqzx7ztnw0000gn/T/arduino_cache_727838 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.CMSIS.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arduinoOTA.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.openocd.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/Users/lacomj/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -verbose /Users/lacomj/Documents/Arduino/GoProControl_Example/GoProControl_Example.ino
Using board 'mkrwifi1010' from platform in folder: /Users/lacomj/Library/Arduino15/packages/arduino/hardware/samd/1.8.6
Using core 'arduino' from platform in folder: /Users/lacomj/Library/Arduino15/packages/arduino/hardware/samd/1.8.6
Detecting libraries used...
/Users/lacomj/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21 [... cut ...]
Using library GoProControl at version 1.2.0 in folder: /Users/lacomj/Documents/Arduino/libraries/GoProControl 
Using library WiFiNINA at version 1.5.0 in folder: /Users/lacomj/Documents/Arduino/libraries/WiFiNINA 
Using library Utilities at version 0.4.6 in folder: /Users/lacomj/Documents/Arduino/libraries/Utilities 
Using library SPI at version 1.0 in folder: /Users/lacomj/Library/Arduino15/packages/arduino/hardware/samd/1.8.6/libraries/SPI 
exit status 1
invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]

lacomj, 2020-06-26:
For what its worth, I also get the same error when I try hardcoding the other camera type names into the main code.  For example:
`GoProControl gp("GP24907984", "cHS-Yy9-9Gv", HERO4);`
So its not just related to `HERO8` type camera being specified as argument 3.

aster94, 2020-06-29:
@lacomj I think I fixed the problem, looks like the compiler for MKR is more finicky than others

[image]

may you try the 1.2.1?

lacomj, 2020-07-01:
Unfortunately, it looks like the problem is still there for me.

I.e., I see the same problem with the 1.2.1 version of the library (see output below).  I assume that you got it to compile on your end though (if I follow the image you posted above).  I will see if I can get a fresh copy of the IDE installed onto another computer (a Windows PC, rather than a mac), and see if that makes a difference.  I am wondering if something is up with my installation of the IDE and compiler.  Maybe a MacOS version of the compiler perhaps.   **I also note that it does correctly compile (no errors) if I select the MEGA2560 for the board.**  Cant test that though, because I don't have a WiFi shield for the those boards.  Only have Wifi for the three SAMD21 boards I mentioned having problems with earlier.

`Arduino: 1.8.13 (Mac OS X), Board: "Arduino MKR WiFi 1010"
GoProControl_Example:14:47: error: invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]
 GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA);
                                               ^
In file included from /Users/lacomj/Documents/Arduino/GoProControl_Example/GoProControl_Example.ino:5:0:
/Users/lacomj/Documents/Arduino/libraries/GoProControl/src/GoProControl.h:76:3: note:   initializing argument 3 of 'GoProControl::GoProControl(const char*, const char*, uint8_t, const uint8_t*, const char*)'
   GoProControl(const char *ssid,
   ^~~~~~~~~~~~
exit status 1
invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

aster94, 2020-07-01:
Can you post your `secrets.h` (changing the said and password if you wish)?

lacomj, 2020-07-02:
Update:
Ok, I tried a fresh install of the IDE onto a different machine and OS (Windows), using 1.2.1 with MKR 1010, but I still get the same error as I got on my Mac.

My `secrets.h` file is below.  The example included with the library is missing the `CAMERA` definition, so I added it in.

`#ifndef SECRETS_H
#define SECRETS_H

// Replace the following:
#define GOPRO_SSID "GP24907984"
#define GOPRO_PASS "cHS-Yy9-9Gv"
#define CAMERA "HERO8"

#endif
`

aster94, 2020-07-02:
Remove the `"`

closed 2020-07-02 by aster94

lacomj, 2020-07-03:
Damn.  Missed that.  Thanks, it is now compiling and I have some control over the GoPro.  Great stuff!  Thanks!