aster94/GoProControl #30: Suggestion that could solve some problems

sepp89117, 2020-12-14:
I have a solution that works as an access point and can control several cameras at the same time. Maybe interested in you?
[https://github.com/sepp89117/GoProRC_ESP32](url)

I would be happy if you had a solution for my case, namely I want to keep the cameras in AP mode switched on while I'm in AP mode myself. An idea?

greetings from Germany

aster94, 2020-12-15:
Hi, AFAIK switching between wifi modes with ESP isn't bug free, maybe you should direct this question to the ESP forum
Feel free to try this library and tell me if this solved your problem

sepp89117, 2020-12-16:
Hi, I made myself unclear. I only use the ESP32 in AP mode. The cameras connect to it and I can control them. That is the principle of a GoPro Smart Remote.
With the computer and software (GoEasyPro in my Repos), I control the ESP using serial communication. At the same time I can connect the computer to the AP of the cameras (only individually) and then access and watch the videos. But the cameras turn off AP mode and I can't turn it back on. Only restarting the cameras will help.
I thought you might have the necessary information to give a tip on how to keep the cameras AP alive.
The Smart Remote works without any problems.

aster94, 2020-12-16:
hello, I send them a request once every few seconds to keep them alive

sepp89117, 2020-12-16:
Yes, I saw that. _GPHD _....
But you send it as a station to the access point. Unfortunately, it doesn't work the other way around for me.

sepp89117, 2020-12-30:
It is not possible to keep the access point of the camera alive with the smart remote. The only way is to switch the cameras off and on again using the smart remote. To do this, send a pw0 and then restart the access point of the smart remote.

closed 2020-12-30 by sepp89117