aster94/GoProControl #33: MultiCam issue with 2x GoPro HERO7 Black

Alquanter, 2021-04-19:
Hi,
I'm trying to control multiple GoPro HERO7 cameras simultaneously. When launching the MultiCam example on my Arduino Wifi Rev2 it connects to the first listed camera but gives an error for the second camera: "Connection failed with status: 6".
When I switch the order of the 2 cameras it again only connects to the first camera and pops up the same error for the second camera.
Do you have an idea on what could be causing this problem?

aster94, 2021-04-19:
@KonradIT are you aware of changes in how newer gopro cameras handle this?

KonradIT, 2021-04-19:
Hi, weird, i wrote this and tested using my HERO4 and HERO7 and it worked. I don't have multiple 7 cameras but I'll try Hero7 and Hero5.

Alquanter, 2021-04-19:
Thanks for your quick reply!
Sequentially 'beginning' and 'ending' a session seems to work fine but 'beginning' a second session without 'ending' the first session throws an error. Are you sure the Arduino can open 2 sessions in parallel on different GoPro's (being different WiFi networks)?

damiendon, 2021-05-19:
I had similar issue with one cam. My esp was returning the same error.
Try to modify the MAX_WAIT_TIME in settings.h at 5000 or more, it worked for me ...

aster94, 2021-05-19:
> Try to modify the MAX_WAIT_TIME in settings.h at 5000 or more, it worked for me ...

I will edit this in the next release!

closed 2021-09-04 by aster94