aster94/GoProControl #35: getstatus and getmedialist gopro Hero3

damiendon, 2021-05-19:
Hi, I'm facing a problem. 

I try to get status and media list on my gopro hero3 with ESP8266.

I tried your arduino json gopro  and for both it return "EmptyInput".

On Konrad documentation, it says that for hero3 the status is given in bytes returned by http://10.5.5.9/camera/sx?t=PASSWORD
I can obtain a file named sx when i do that in my browser, and this file contain bytes for decoding the status.
Here the library looks only to a json, so i think it cant work for my hero3 (with status).

for the media list, your exemple arduinojsongopro should work because for hero3 the media list is a json.

So I think "getstatus" and "getmedialist" are not returning the values returned by http://10.5.5.9/camera/sx?t=PASSWORD and http://10.5.5.9:8080/gp/gpMediaList

How can I fix that ?

Thank you ! :)

aster94, 2021-05-19:
okkkk I wasn't aware of this difference, you should change the example:
https://github.com/aster94/GoProControl/blob/75f3c63532ba92eb5b16f05e8e5362a4150af390/examples/ArduinoJson/ArduinoJson.ino#L19

Here change this line and just print it and delete the json stuff

damiendon, 2021-05-19:
Thinking about it today, maybe the buffer is not big enough...

for status, empty input is normal as the deserializeJson doesn't find any json object. Alsow, when i try to print to serial gp.getstatus(), it only return request reponse stuff but no bytes... for that I don't Know what to do

For Media list it should work, but it alsow return empty input... thats make me think that buffer is to small and it stop recording then entry when full, so deserializejson don't find the last part of the Json object. Alsow, when i just try to print in serial gp.getmedialist(), it only return request reponse stuff but no json object...

damiendon, 2021-05-21:
OK,
For Status i find a way to read the bytes (without the library), i will try to make a working function for the library when I got time.
for now with the library, if I just print gp.getStatus(), I don't have the bytes of the status, I don't know why...

For MediaList, I still can't see the json, but I think it's because the rx buffer when the json is returned is not big enough due to limitiations with esp8266 memory. I have about 150 jpg and 150 mp4 wich give me a json of about 15000 characters (making tests make me do a lot of pictures and videos ! haha).
reading posts it looks like the buffer size is about 1500 MAX... 
I will give it a try and see if my issue for medialist is solved. I will give the result here, maybe it could help somebody one day.

aster94, 2021-05-21:
Perfect, keep us updated

damiendon, 2021-05-31:
Hi ,
I made a pull request for the medialist part, it's working on my hero3, if somebody can test it on others gopro, I only have a hero3 !
the problem was the port 80, it work on port 8080 for me.

next step with getstatus for hero3 :)

closed 2021-09-04 by aster94