aster94/GoProControl #40: Issue with connecting/shooting photos after 1st time ESP8266 & Hero 7

PlastiBots, 2021-12-18:
Hi.  Have a challenge with failed connections after the first attempt.   I can only get the ESP to trigger picture captures after resetting the GoPro WiFi connection and rebooting it.  The first time in, I can connect and take photos.  After this, it fails. Regardless of power cycling the ESP or the GoPro   Below is the code I am using, and further down is the Serial output:
`
#include <GoProControl.h>
#include "Secrets.h"

/*
  Control your GoPro with the Serial Monitor
  edit the file Secrets.h with your camera netword name and password
  CAMERA could be: HERO3, HERO4, HERO5, HERO6, HERO7, FUSION, HERO8, MAX
*/

#define CAMERA HERO7 // Change here for your camera
//from my Hero 7:  06 41 69 EF 79 94
byte remote_MAC_ADD[] = { 0x06, 0x41, 0x69, 0xEF, 0x79, 0x94 };
const char* board = "NodeMCU";   //ESP Board Name

GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA, remote_MAC_ADD, board);

void setup()
{
  gp.enableDebug(&Serial);

  while (!gp.isConnected())
  {
    gp.begin();
  }
  
  for (int i=0; i<5; i++)
  {
    Serial.println("Attempting a shot.");
    gp.shoot();
    gp.keepAlive(); // not needed on HERO3
    delay(1000);
  }  
}

void loop()
{
  Serial.println("In Main Loop. Keep alive.");
  gp.keepAlive(); // not needed on HERO3 
}
`

Attempting to connect to SSID: "GP50039006"
Using password: "kS9-mMx-Rvx"
.......................................................
Connected to GoPro
Attempting a shot.
Connection lost
Attempting a shot.
Connect the camera first
Attempting a shot.
Connect the camera first
Attempting a shot.
Connect the camera first..

aster94, 2021-12-19:
Have you tried the example in the library, does it work?

PlastiBots, 2021-12-19:
Hi.  Yes, I tried the example and the same thing.  It only ever works the first time connected, then after it fails.  Only way it works again is by resetting the wifi connection on the GoPro, then it works again for the first time only and fails.  I've read other threads with a similar issue.  It's as if the GoPro is closing the connection.

aster94, 2021-12-19:
I think your delay is too long, you need to call the keep alive more often

PlastiBots, 2021-12-19:
I tried that. If you look at the sample I included, it dives right into attempting to take a shot while also having a keepAlive() in there.  I've even removed it from setup() and tried it in the loop.  All result in the same thing. Works fine the first time, then fails until GoPro WiFi is reset.

PlastiBots, 2021-12-21:
So, I figured out the steps necessary to cause the fail, but still no closer to determining why.  This has only been validated on a Hero 7 Black.  After doing all sorts of tests including restarts (3s and 10s), removing power from the GP,  downgrading the GP FW to 1.6.1 (was 1.8.x) it fails the same way every time.  The only success I could get is to ensure that I always called gp.end() before powering down the ESP or camera. I was able to test this by calling it, then powering down the ESP and GP in different sequences. It always worked when powered back up.  If either is powered down without this call, it fails and becomes permanent.  So, far it seems that a complete reset of the GP is necessary.    Code for this is below. I'm using a M5Stack atom, but any ESP32 with a few buttons wired up will do.  In this case, a single click takes a shot, while holding it > 2 seconds calls gp.end().  A bit of a pain but it works.  What it also means as I need to device a power disconnect circuit which feeds into a software call go gp.end().  Hoping I can do this with a supercap and some other bits but not an ideal soln.  Although I'm not certain, I think the GoPro is failing with some feature where it hasn't let go of a setting or mode.  I'm thinking something related to the UDP client and maybe flushing things.    It seems that only a full WiFi reset (reset connections) will fix it if you fail to call gp.end() before powering down the ESP...  sigh...  

```
#define CAMERA HERO7 // Change here for your camera
//from my Hero 7:  06 41 69 EF 79 94
byte remote_MAC_ADD[] = { 0x06, 0x41, 0x69, 0xEF, 0x79, 0x94 };
const char* board = "RTLapseCAM";   //ESP Board Name

GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA, remote_MAC_ADD, board);

void setup()
{
  gp.enableDebug(&Serial);
  //xTaskCreate(keep_alive, "keep_alive", 10000, NULL, 1, NULL);  //for ESP32

  while (!gp.isConnected())
  {
    gp.begin();
  }
  //delay(1000);
  //gp.turnOn();
  //gp.confirmPairing();
  //gp.keepAlive();
  //gp.setMode(PHOTO_MODE);
  //gp.keepAlive();
  //delay(1000);
  //gp.shoot();
  //gp.keepAlive();
  //delay(1000);
  //gp.shoot();
  //delay(1000);
  //gp.shoot();
  Serial.println(">>        Printing Status...");
  gp.printStatus();
  delay(1000);

 
}

void loop()
{
  M5.update();    //Read the press state of the key. 
  
  if (M5.Btn.wasPressed())
  {
    Serial.println("Button - take a shot.");
    //if (!gp.isConnected()){gp.begin(); delay(2000);}
    gp.shoot();
  }

  if (M5.Btn.pressedFor(2000))  //pressed for 2 seconds or more.
  {
    Serial.println("Closing connection.");
    gp.end();
  }

  //Serial.println("In Main Loop. Keep alive.");
  gp.keepAlive(); // not needed on HERO3
  delay(10);
}
```