aster94/GoProControl #41: add command to download images

morray, 2021-12-30:
Could have additional options, like
a) last image
b) first image
c) all images
d) specific images
e) direct transfer to google/AWS/one/... cloud
f) direct transfer to ftp/smb/....

KonradIT, 2021-12-30:
Downloading >5MB files in an ESP32? Sounds pretty challenging.

You could use https://github.com/aster94/GoProControl/blob/master/examples/ArduinoJson/ArduinoJson.ino#L47 to get a deserialized JSON object of the media list. And from there use something like [this](https://www.iotsharing.com/2018/01/esp32-multipart-upload-file-and-download-via-http.html) to download the file to an SD card (construct your URL like: `http://10.5.5.9:8080/videos/DCIM/$folder/$file.JPG`).

As per e) and f)... that's pretty much out of scope.

aster94, 2021-12-31:
That library seems interesting and not so heavy so it could be implemented into a download function
Agree with @KonradIT e) and f) are out of scope