aster94/GoProControl #43: Getting "Connection lost" quite a lot

beckmx, 2023-02-02:
Hello guys, I was giving a try to the library, initially i was able to execute all the commands and then from one day to the other without modifying anything I am getting a lot of connection lost, I am using an ESP32 and what I do notice is that if I connect to it via simple Wifi.begin() it does connect but any of the commands respond, they all get "rejected" or not answered, I am using a Hero7 black, if I connect my computer to the camera I am able to access the different APIs, but not on the esp32, I was wondering if there are new headers or something new, I even tested with a more simple script that just pulls the status but on the esp32 seems not to work consistently.

```
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "Unknown HERO7 Black";
const char* password = "snv-GRX";

String serverName = "http://10.5.5.9:8080/gp/gpMediaList";

unsigned long timerDelay = 7000;

void setup() {
  Serial.begin(115200); 

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
  if ((millis() - lastTime) > timerDelay) {
    if(WiFi.status()== WL_CONNECTED){
      HTTPClient http;

      String serverPath = serverName;
     
      http.begin(serverPath.c_str());

      int httpResponseCode = http.GET();
      
      if (httpResponseCode>0) {
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
        String payload = http.getString();
        Serial.println(payload);
      }
      else {
        Serial.print("Error code: ");
        Serial.println(httpResponseCode);
      }
      // Free resources
      http.end();
    }
    else {
      Serial.println("WiFi Disconnected");
    }
    lastTime = millis();
  }
}
```

aster94, 2023-02-02:
Unfortunately it's true the link between esp32 and GoPro is not so stable,
I don't know the reason.
For sure try to always close the connection, using also error handling,
this will improve the connection a lot

PlastiBots, 2023-02-02:
I did 2 projects using GoProControl. Managed to get it to a point where the ESP was stable with my Hero.    This [one is the first ](http://www.plastibots.com/index.php/2021/12/25/rtlapsecam-resin-printer-time-lapse-with-m5stick-c-plus-gopro/)with code. The [second one](http://www.plastibots.com/index.php/2022/02/02/rtlapsecam-resin-printer-time-lapse-with-m5stick-c-plus-gopro-2/) added a slider to the setup, but similar code.  The ESP connection bits should be the same between the two. Sorry, I did this months ago so don't recall the inner workings but the sketch should be self explanatory.

beckmx, 2023-02-02:
@PlastiBots I just noticed your comment in the code:

> There's an issue somewhere between comms of the ESP and GoPro.  This took me a week to figure out.  Turns out
>   that if you disconnect / power down the ESP or GoPro without first successfully calling gp.end() something 
>   gets hung in the comms.  The ESP will still connect to the GoPro but it will return connection lost errors. 
>   To fix this, simply disconnect ESP power, turn off GoPro and remove battery (I did for a few mins, but it may  take less)
>   Then power up the GoPro, then the ESP and try again. I think it also works if you just remove the GoPor
>   battery, wait a bit, then power it back on and re-connect.  

And I also noticed you set up the Mac Address of the camera, I am going to follow on that as well, thanx!

closed 2023-02-02 by beckmx

beckmx, 2023-02-03:
@PlastiBots @aster94 a word of advice guys, I ran this library inside the Raspberry pico w, and duuude, I mean duuude, what a blast! it's so incredibly fast, the only changed I did to make it work is in the file:
`GoProControl.h`
line 30 I added this:
`#if defined(ARDUINO_ARCH_ESP32) || defined(ARDUINO_ARCH_MBED_RP2040)|| defined(ARDUINO_ARCH_RP2040)  // ESP32`
The board manager I used was from this repo https://github.com/earlephilhower/arduino-pico, the rest of the code was fully compatible and the connection was so stable I didn't have to worry anymore

aster94, 2023-02-05:
thanks for the info @beckmx