aster94/GoProControl #44: Issue compiling RTLapseCAM

Halcath, 2023-02-21:
I am fairly new to coding etc. Was just hoping to get this old gopro4 silver to take timelapse of my resin printers.

When I try and compile the sketch in Arduino IDE 2.0.3 I get the following error

Compilation error: 'HERO4' was not declared in this scope

Pointing to this line:

#define CAMERA HERO4 // Change here for your camera
//from my Hero 7:  06 41 69 EF 79 94
byte remote_MAC_ADD[] = { 0x06, 0x41, 0x69, 0xEF, 0x79, 0x94 };
const char* board = "RTLapseCAM";   //ESP Board Name

Also, since I am worried it will come up.....do I need to find my cameras mac address and also insert that? According to the gopro memory card the mac add is "d4d919c91faa"

aster94, 2023-02-21:
Change that line to match your GoPro model
Change the Mac address splitting two by two

Halcath, 2023-02-21:
Thats what I have done, but I get the same error

"Compilation error: 'HERO4' was not declared in this scope"

#define LDRSensorPIN 33    //was 26 but now working??
#define BUZZER_PIN 0
#define BUZZER_CHANNEL 0

#define CAMERA HERO4 // Change here for your camera
//from my Hero 7:  d4 d9 19 c9 1f aa
byte remote_MAC_ADD[] = { 0xd4, 0xd9, 0x19, 0xc9, 0x1f, 0xaa };
const char* board = "RTLapseCAM";   //ESP Board Name

Halcath, 2023-02-21:
I get the same error if I try to compile your example sketch
"Compilation error: 'HERO3' was not declared in this scope"
Screenshot here: https://postimg.cc/rRw87HfZ

Halcath, 2023-02-22:
I tried asking chatgpt to troubleshoot the sketch. After giving it just the sketch it said the syntax looked fine. I then tried telling it the error I get when compiling. It suggested adding quotation marks around HERO3 so I tried "HERO3". When trying to compile that it started  giving a bunch of new errors. 

I did notice, after I try to compile it without any modification and I hover over 
GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA);
I get 
variable gp
Type: class GoProControl

GoProControl gp("xxxxxxxxx", "xxxxxxx", <recovery - expr>())

network name and password are correct from secret.h  
is the <recovery - expr>     what is suppose to be there?

If I take chat GPT's suggestion and add quotes around HERO3 then instead of <recovery - expr>()  It does say HERO3 there but as I said, then I get a bunch of other errors.

Halcath, 2023-02-22:
One more note. 
I emailed the person that wrote RTLapse yesterday.
He just responded and said
"Odd. I tried to compile my last version and got the same error for my HERO7 - which was working fine months ago.  I'm as baffled as you are.  It should work, "

He then suggested I ask here........ 

So apparently it isn't just me.

I have also run back through all the available versions of goprocontrol inside arduino IDE and get the same error with every version except 0.9.3 which just fails as soon as it starts trying to compile.
I have also tried going back to a previous version of Arduino IDE (1.8.18) that I still had on my PC. It gives the same error.
At a total loss here and don't know what else to try or who to ask.

aster94, 2023-02-23:
try the version 1.3.1 of the library

Halcath, 2023-02-23:
I have tried the example goprocontrol.ino 
The person from plastibots is getting the same error
5 different people from discord have all tried it and gotten the same error
The web developer that works for my wife at the university gets the same error
I have found zero people that can get the sketch to compile
[image]

Halcath, 2023-02-23:
OMG I found the issue!!!!!!!!!!!
Its the M5stack library!!!!
I saw you had a fix for compile issues on some other board so I figured I had tried everything else, I'll try rolling back the M5stack library.
Newest version is 2.0.6 which fails, as does the previous version
Rolling M5Stack board library back to 2.0.4 WORKS!!!!!
Compiled and installed!!!!!

Halcath, 2023-02-23:
yeah sorry, not library, in my excitement after days of frustration I wrote that wrong.
But yes rolling back the M5stack board to V 2.0.4 is what finally worked for me.
 I am on:
  V 1.3.1 of the goprocontrol library
  V 2.0.4 of the M5stack board
  
  Compiled and installed.
  Connect to my gopro first try and is taking pics as it should when manually blocking and exposing the photoresistor
  Now to mount it and run a print!!!!
[image]

Halcath, 2023-02-23:
Huge thanks to Dave at PlastiBots and to Aster/goprocontrol 
Sorry for all the endless comments, was just trying to document everything I tried.
goprocontrol v1.3.1 (library)
M5stack v2.0.4 (board manager)
Seems to be the solution.....at least for me

aster94, 2023-02-23:
I am happy it worked but honestly I don't understand why you got that error 🤷🏻‍♂️
Anyway I tried to compile with a mega board, and also the 1.3.2 should work

closed 2023-02-23 by aster94

Halcath, 2023-02-23:
Can confirm
Goprocontrol 1.3.2 does work.

There is something in the latest two versions of M5stack that cause the error which would explain why you were not getting the error when compiling for other boards.
Which is weird because it errors on a portion (camera enumeration) that is handled by goprocontrol.
But rolling back to 2.0.4 seems to solve the issue. 
Again, thank you very much!