aster94/GoProControl #45: Problems trying to connect to MAX...

paicolman, 2023-02-25:
Hi, 
Thanks for this library, I had written a remote for my Gopro MAX but stopped using it for a while and now it does not connect anymore, it always fail with code 6 or 5. Thought I try your example but got stuck in the same place, it just won't connect... SSID and PWD are correct, but the camera just ignores any connection. Thought I ask if maybe you know of any new trick needed by Gopro. I suspect you would need to go through Bluetooth first... 

Thanks a lot!

aster94, 2023-02-26:
hello, yes, as you suspect it needs to connect to ble first, i use the mobile app to activate it

closed 2023-02-26 by aster94

paicolman, 2023-02-26:
Yes that used to work before, but not anymore. When I connect with the newest version of the App, it seems to block it somehow. If I press "view media" and then start loggiing in, it tryies to work (connected to Gopro) but then it falls out immediately. I have been trying to hack the BLE connection by myself with the ESP, but no luck so far. Thanks a lot anyway, your tool will be useful once I solve the starting issue!!

aster94, 2023-02-26:
It is also possible that something changed in the GoPro software so, as you noted, the old trick doesn't work anymore
Honestly I don't have a GoPro anymore but having the 
esp32 starting the ble connection would be a great improvement