aster94/GoProControl #47: Format needed for Secrets.h

Gerald-D, 2023-09-26:
Hi
I am very new to Github, Arduino and WiFi. Sorry in advance for some some of the stupid questions I may be asking. . . . . 

I make YouTubes with up to 3 GoPros (1xHERO6 and 2xHERO8). See channel https://www.youtube.com/channel/UCyqo4XTQlz57-ESk47V6tCg The cameras must shoot video for 3 seconds, wait 57 seconds and then repeat cycle. I manage fine if there is only one camera in front window (using a hacked Yoctop remote control). But when there are also cameras on the side windows, they often go out of sync and record for 57 seconds - that chews up precious SD card space. Therefore I need to address commands individually to each of the cameras, according to their status. I believe that I can do it with an ESP32 and a lot of learning from guys like you on Github. Thank you in advance.

My first problem is with Secrets.h:

// replace the following:
#define GOPRO_1_SSID "__YOUR_CAMERA_NAME__"
#define GOPRO_1_PASS "__YOUR_CAMERA_PASS__"

// same for the second camera:
#define GOPRO_2_SSID "__YOUR_CAMERA_NAME__"
#define GOPRO_2_PASS "__YOUR_CAMERA_PASS__"

If I look at GoPro Quik on my Android, I see for one of the Hero8's
Name: GP53089464  
Password: C>T-BNB-p4Q
(I scrambled them slightly, but correct length and type of characters)

Is it these names and passwords? Must it look like this?
#define GOPRO_1_SSID "__GP53089464__"
#define GOPRO_1_PASS "__C>T-BNB-p4Q__"

aster94, 2023-09-26:
Hello,

Yes the format is correct, the wifi is very stable if you connect to one
camera. The problems will arise if you try to connect and disconnect to two
or more cameras. You are lucky because you have 57 seconds to disconnect
and reconnect so I guess you will be able to complete your project

Gerald-D, 2023-09-26:
Thank you very much for the quick reply.

What I didn't tell you yet, is that I have a manual trigger button near my car's gear lever - I push it when I see an interesting scene and want to force an immediate recording. Already I have learnt that the Hero8 takes about 4 seconds to save a file and does not want another "record" instruction in that time. All my problems of cameras going out of sync have so far been after pushing the manual button. However, I have good faith that with the ESP32 and proper WiFi feedback, that I will be able to program a more intelligent solution.

PlastiBots, 2023-09-26:
FYI. You may find [my write-up](http://www.plastibots.com/index.php/2021/12/25/rtlapsecam-resin-printer-time-lapse-with-m5stick-c-plus-gopro/) useful to avoid some pitfalls I discovered while trying to use an ESP to control my Hero 7 for timelapse.  Specifically this section:  "_Onto GoProControl which connects to the GoPro’s own WiFi"_

Gerald-D, 2023-09-26:
> FYI. You may find [my write-up](http://www.plastibots.com/index.php/2021/12/25/rtlapsecam-resin-printer-time-lapse-with-m5stick-c-plus-gopro/) useful to avoid some pitfalls I discovered while trying [...]

Thank you for that. I will certainly study it.

closed 2023-09-27 by Gerald-D