aster94/GoProControl #48: Stuck on line 10 with a HERO8

Gerald-D, 2023-09-27:
I have this as line 10 of GoProControl.ino :
#define CAMERA HERO8 // Change here for your camera

And then the first error in the compiler is:
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:10:16: error: 'HERO8' was not declared in this scope
 #define CAMERA HERO8 // Change here for your camera
 
 I tried HERO6 as well, but still similar error message
 
 Obviously I need to learn some more very basic things . . . . . .

Gerald-D, 2023-09-27:
In my application, I will probably develop on your MultiCam.ino, because all the camera settings are already done when switching on and connecting. (In fact the camera(s) are left switched on and wifi connected for days at a time).

MultiCam.ino also has me stumped

aster94, 2023-09-28:
see https://github.com/aster94/GoProControl/issues/46

Gerald-D, 2023-09-28:
#46 helped a little . . . . . I am now stuck at line 12:

C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:12:47: error: invalid conversion from 'const char*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]
 GoProControl gp(GOPRO_SSID, GOPRO_PASS, CAMERA);
                                               ^
In file included from C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:1:
C:\Users\Gerald\Documents\Arduino\libraries\GoProControl\src/GoProControl.h:82:30: note:   initializing argument 3 of 'GoProControl::GoProControl(const char*, const char*, uint8_t, const uint8_t*, const char*)'
                const uint8_t camera,
                ~~~~~~~~~~~~~~^~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino: In function 'void loop()':
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:48:19: error: 'HERO3' was not declared in this scope
     if (CAMERA == HERO3)
                   ^~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:110:16: error: 'VIDEO_MODE' was not declared in this scope
     gp.setMode(VIDEO_MODE);
                ^~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:110:16: note: suggested alternative: 'SIGEV_NONE'
     gp.setMode(VIDEO_MODE);
                ^~~~~~~~~~
                SIGEV_NONE
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:114:16: error: 'PHOTO_MODE' was not declared in this scope
     gp.setMode(PHOTO_MODE);
                ^~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:114:16: note: suggested alternative: 'RMT_TX_MODE'
     gp.setMode(PHOTO_MODE);
                ^~~~~~~~~~
                RMT_TX_MODE
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:118:16: error: 'MULTISHOT_MODE' was not declared in this scope
     gp.setMode(MULTISHOT_MODE);
                ^~~~~~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:123:23: error: 'ORIENTATION_UP' was not declared in this scope
     gp.setOrientation(ORIENTATION_UP);
                       ^~~~~~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:127:23: error: 'ORIENTATION_DOWN' was not declared in this scope
     gp.setOrientation(ORIENTATION_DOWN);
                       ^~~~~~~~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:132:20: error: 'MEDIUM_FOV' was not declared in this scope
     gp.setVideoFov(MEDIUM_FOV);
                    ^~~~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:136:21: error: 'FR_120' was not declared in this scope
     gp.setFrameRate(FR_120);
                     ^~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:136:21: note: suggested alternative: 'MR_0'
     gp.setFrameRate(FR_120);
                     ^~~~~~
                     MR_0
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:140:27: error: 'VR_1080p' was not declared in this scope
     gp.setVideoResolution(VR_1080p);
                           ^~~~~~~~
C:\Users\Gerald\Documents\Arduino\GoProControl\GoProControl.ino:144:27: error: 'PR_12MP_WIDE' was not declared in this scope
     gp.setPhotoResolution(PR_12MP_WIDE);
                           ^~~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'const char*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]

Gerald-D, 2024-04-16:
Hi Aster

In #46 you said this could be because of an Arduino IDE problem. If I may ask; what IDE did you use?

Thank you

Gerald-D, 2024-04-17:
With some help from a friend, found that:

1. The file names settings.h or Settings.h are used by different libraries. The required file is included by <...> angle brackets in the GoPro Lib. Those brackets usually lead Arduino IDE to a search in the directory where the standard libs are and loads one of those libs instead of the one which is connected to the GoPro lib. In GoProControl.h , changed #include <Settings.h> to #include "Settings.h"

2. In line 409 in GoProControl.cpp there is a line
      int16_t end = start + len;
My compiler throws an error as the variable "end" is not used in its scope.
Temporary Solution: Just comment the line out:
//      int16_t end = start + len;

My friend also suggested to add:
#pragma once
to the Settings.h  It tells the compiler to read this file only once and not everytime if it is called at several places

aster94, 2024-04-18:
Nice, if you wish you can make a PR